i2cy/CH347-HIDAPI #1: I2C

Robo-NK47, 2023-05-05:
Hi guys,

Is there support in I2C communication??

i2cy, 2023-05-06:
Since I do not need i2c communication currently, no. But I may add the support for I2c later since it uses usb hid communication and it should be easy.

获取Outlook for Android<https://aka.ms/AAb9ysg>

Robo-NK47, 2023-05-06:
Thank you for your response :-)

I really need this communication type and don't have the technical abilities to write it at this time.
Do you have any tips? Are you capable of doing it?

i2cy, 2023-05-06:
Sorry I am not available for this update recently, things have been busy lately. I will update the I2C support once I got the spare time.

i2cy, 2023-05-11:
an update of this repositories should probably be committed next mons.

closed 2023-05-11 by i2cy