i2cy/CH347-HIDAPI #7: I2C Repeated Start Conditions

yiziliuxian, 2024-10-31:
I am trying use this library with some TI chip. Could you please explain how to use i2c.read(reg,length) to do **Pointer Byte Set Followed by a Repeat Start and Word (TwoByte) Read** as showing in the following figure? Now I can only use i2c.read and i2c.write seprately.

[image]

i2cy, 2024-10-31:
A typical I2C read situation that resembles reading from mpu6050 chip:

[image]

The **Pointer Byter from Master** byte you mentioned resembles the **Reg Address** byte shown above, and they were both followed by a **Repeat Start** signal from the master device (CH347). And the whole process may break down to two seperated steps: write 1 byte and read couple of bytes (2 bytes in your specific situation).

To perform a **_Repeat Start and Word Read_** (that your document called **_Repeat Start and Word Read_**) using this libraray, the code may look like this:

```
from ch347api import I2CDevice as i2c

dev = i2c(addr=0x48)  # in case of A0=0, A1=0, replace with actual 7-bits address of yours
pointer = 0xff  # replace with your pointer byte
response = dev.read(reg=pointer, length=2)  # Repeat Start and Word Read
```

I hope this would help :)

yiziliuxian, 2024-11-04:
I tired you code but I found it doesn't work. I am currently using the following code:

```
i2c.write(reg=regbit)
d = i2c.read(reg=None, length=2)
```

closed 2025-08-08 by i2cy